dymnz/sEMG #11: ADC interval swing

dymnz, 2017-04-10:
Arduino ADC reading interval swing:
* Come from micros() call?
  * Check the interval of consecutive micros() call
* Effect of interval swing?
  * Ask teacher

LINPOYU, 2017-04-10:
hihi~

dymnz, 2017-04-12:
https://github.com/LINPOYU/ultrasound/issues/3
https://github.com/LINPOYU/ultrasound/issues/4

dymnz, 2017-04-12:
### Probability of time interval between two consecutive ADC read
[image]
### Probability of time interval between two consecutive micros() call
[image]

---
### Conclusion
micros() call cause the ADC read interval swing

closed 2017-04-12 by dymnz

dymnz, 2017-05-08:
### Probability of time interval between two consecutive ADC read with 400us delay
[image]
### Probability of time interval between two consecutive micros() call
[image]

---
### Conclusion
micros() call cause the ADC read interval swing

dymnz, 2017-05-08:
### Probability of time interval between two consecutive ADC read with 400us delay by delaymicroseconds()
[image]
### Probability of time interval between two consecutive micros() call
[image]

---
### Conclusion