dymnz/sEMG #15: Force detection of static signal file on PC

dymnz, 2017-05-06:
Given a sEMG signal, find a way to detect if force is used.

- [ ] Signal files given

dymnz, 2017-05-10:
訊號在這裡 https://github.com/dymnz/sEMG/tree/master/Signals

### Usage
Please refer to [Signal plotting example](https://github.com/dymnz/sEMG/tree/master/ADC_NI_DAQ/Signal%20plotting%20example) for Matlab

dymnz, 2017-05-17:
@abb133aaa111 @max2468tw 

你們星期五下午或星期六有空嗎? 我可以教你們怎麼用新電極和arduino讀取

max2468tw, 2017-05-17:
星期六早上應該方便 可是要meeting
請問星期五晚上可以嗎?還是可能有些晚

dymnz, 2017-05-17:
OK, 幾點?

max2468tw, 2017-05-17:
做完電子實驗6. 假若晚餐後6.半方便嗎? 之後到晚上都沒什麼事!

dymnz, 2017-05-17:
那星期五晚上七點好了

abb133aaa111, 2017-05-17:
我可以

abb133aaa111, 2017-05-19:
我用內建的example blinkwithoutdelay出現這個結果
​

dymnz, 2017-05-20:
@abb133aaa111 
你從email傳的嗎?  看不到你的結果

abb133aaa111, 2017-05-21:
對 那這樣呢
http://i.imgur.com/KdnjOpm.png
內建的example blinkwithoutdelay

dymnz, 2017-05-21:
這是有成功燒進去了. 只是你沒有改led pin, 所以看不出結果.

closed 2017-07-19 by dymnz